arduino-libraries/ArduinoECCX08 #3: Compile error ECCX08RandomNumber.ino

Skysurfer-14, 2018-11-12:
On compiling the example file in Platformio, I get an error:

C:\Users\user\.platformio\lib\ArduinoECCX08_ID5570\src\ECCX08.cpp:622:80: error: call of overloaded 'requestFrom(uint8_t, size_t, bool)' is ambiguous

It seems, that changing the Line 622 of ECCX08.cpp to
`while (_wire->requestFrom((uint8_t)_address, (size_t)responseSize) != responseSize && retries--);`
solves the problem.

sandeepmistry, 2018-11-20:
Hi @Skysurfer-14,

Could you please provide more details on you setup? Like which board you are targeting and full verbose output.

sandeepmistry, 2018-11-20:
Sorry, I can reproduce it when the Uno is selected, I just saw your edited comment.

This library has only been tested on SAMD boards, like the Arduino MKR series.

sandeepmistry, 2018-11-21:
Closing this as won't fix, based on the comment in https://github.com/arduino-libraries/ArduinoECCX08/issues/4#issuecomment-440718620.

closed 2018-11-21 by sandeepmistry