arduino-libraries/ArduinoECCX08 #4: Buffer overflow in wire.requestFrom()

Skysurfer-14, 2018-11-13:
The example script ECCX08RandomNumber.ino did not work correctly. With a logic analyzer on the I2C I can see, that the random number ist transmitted from ATECC508A, but 20 times. It seems, that the buffersize of the wire.h lib can only receive 32 byte and the ATECC sends 32 + 3 byte (length and CRC) byte. This error appears in 
`int ECCX08Class::receiveResponse(void* response, size_t length)`
and leads to a CRC-error.

sandeepmistry, 2018-11-20:
Hi @Skysurfer-14,

Is this like #3, and you are building for the Uno again?

Skysurfer-14, 2018-11-21:
Hi @sandeepmistry ,
yes, I get this error on the Uno. It's nearly working, as described above, but receiving the data from ATECC508A did not work correct.

sandeepmistry, 2018-11-21:
After discussing this with @facchinm, we've decided to not support AVR boards like the Uno. As it hasn't been tested and we are seeing issues like #3 and this one.

The next release of the library will limit the architectures the library is available on to Arduino SAMD and megaAVR boards. This change was made in pull request #5.

closed 2018-11-21 by sandeepmistry

Skysurfer-14, 2018-11-22:
Hi @sandeepmistry,
In the meantime I found out, that increasing the BUFFERSIZE in wire.h AND twi.h to 70 solves the problem (together with Issue #3 ). With this changes the example script ECCX08RandomNumber.ino works also on UNO!