arduino-libraries/ArduinoECCX08 #18: Initialisation failure when using TWI pins

ubidefeo, 2019-05-08:
A user in the forums had issues which were tracked back to this library's begin.
When using a motor shield via a MKR to UNO adapter, pins 11 and 12 were interfering with the correct functioning of the TWI.
`begin()` could use a timeout and return failure.
This has repercussions on `ArduinoIoTCloud`.
Forum post here
https://forum.arduino.cc/index.php?topic=606191.new#new

sandeepmistry, 2019-05-13:
@ubidefeo do you have the h/w to reproduce?

Is if failing or hanging? Begin already returns 0 on failure, can you confirm this is the case?

ubidefeo, 2019-05-14:
@sandeepmistry 
it hangs, check the forum post linked above.
It's an i2c issue

sandeepmistry, 2019-05-14:
@ubidefeo if it's an I2C issue, I suggest you open an issue in the SAMD core instead: https://github.com/arduino/ArduinoCore-samd/issues

Please let us know the state of the SCL/SDA pins on the MKR, are they tied LOW or HIGH etc.

ubidefeo, 2019-05-18:
@sandeepmistry , sorry for the late reply.
been crunching hard for Maker Faire :)
I don't have one of those shields, but I plan on having one shipped to me.
Once I test this I'll let you know :)

sandeepmistry, 2019-09-18:
Hi @ubidefeo,

I'm going to close this now due to lack of feedback. Please re-open with more info once your get a chance to test.

closed 2019-09-18 by sandeepmistry