arduino-libraries/ArduinoECCX08 #25: Support for ESP32

fabltd, 2020-07-10:
Hi

This will not compile on the ESP32 due to differences in the wire lib.

Please fix.

Bach-Danielsson, 2020-08-10:
Hi 
In the ECCX08.cpp file change from
while (_wire->requestFrom((uint8_t)_address, (size_t)responseSize, (bool)true) != responseSize && retries--);
 to
while (_wire->requestFrom((uint8_t)_address, (uint8_t)responseSize, (bool)true) != responseSize && retries--);

then you dont get the comment from your compler regarding differences in the wire lib,

fabltd, 2020-08-10:
Hi

Still a lot of warnings:

lib/ATECCx08/src/ECCX08.cpp: In member function 'int ECCX08Class::receiveResponse(void*, size_t)':
lib/ATECCx08/src/ECCX08.cpp:724:81: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
   while (_wire->requestFrom((uint8_t)_address, (uint8_t)responseSize, (bool)true) != responseSize && retries--);
                                                                                 ^
In file included from lib/ATECCx08/src/ECCX08.h:24:0,
                 from lib/ATECCx08/src/ECCX08.cpp:22:

/Users/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:98:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int, int)
     uint8_t requestFrom(int address, int size, int sendStop);
             ^
/Users/jeremy/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:96:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)
     uint8_t requestFrom(uint8_t address, uint8_t size, uint8_t sendStop);
             ^
lib/ATECCx08/src/ECCX08.cpp:724:81: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
   while (_wire->requestFrom((uint8_t)_address, (uint8_t)responseSize, (bool)true) != responseSize && retries--);
                                                                                 ^
In file included from lib/ATECCx08/src/ECCX08.h:24:0,
                 from lib/ATECCx08/src/ECCX08.cpp:22:

/Users/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:98:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int, int)
     uint8_t requestFrom(int address, int size, int sendStop);
             ^
/Users/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:94:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint16_t, uint8_t, uint8_t)
     uint8_t requestFrom(uint16_t address, uint8_t size, uint8_t sendStop);
             ^
lib/ATECCx08/src/ECCX08.cpp:724:81: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
   while (_wire->requestFrom((uint8_t)_address, (uint8_t)responseSize, (bool)true) != responseSize && retries--);
                                                                                 ^
In file included from lib/ATECCx08/src/ECCX08.h:24:0,
                 from lib/ATECCx08/src/ECCX08.cpp:22:
/Users/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:98:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int, int)
     uint8_t requestFrom(int address, int size, int sendStop);
             ^
/Users/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:93:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint16_t, uint8_t, bool)
     uint8_t requestFrom(uint16_t address, uint8_t size, bool sendStop);

torntrousers, 2021-02-16:
That [change suggested](https://github.com/arduino-libraries/ArduinoECCX08/issues/25#issuecomment-671357303) by @Bach-Danielsson works for me on an [M5Stack IoT EduKit](https://m5stack.com/products/m5stack-core2-esp32-iot-development-kit-for-aws-iot-edukit). Would be great to get it out in a release of the library.