arduino-libraries/ArduinoECCX08 #41: Self-signed cert generation errors out on some slots

scarolan, 2022-07-26:
I'm using this on an ESP32 device (M5Stack Core2 for AWS).  The error message isn't very specific.  Here's what I get trying to use slot 0 and storage slot 8:

```
Please enter the issue year of the certificate? (2000 - 2031) [2022]:
Please enter the issue month of the certificate? (1 - 12) [1]:
Please enter the issue day of the certificate? (1 - 31) [1]:
Please enter the issue hour of the certificate? (0 - 23) [0]:
Please enter how many years the certificate is valid for? (1 - 31) [1]:
What slot would you like to use for the private key? (0 - 4) [2]: 0
What slot would you like to use for storage? (8 - 15) [8]: 8
Would you like to generate a new private key? (Y/n) [Y]: Y

Error starting self signed cert generation!
```

I was able to eventually get it working with Slot 2, Slot 8 but the "generate new private key" option never worked for me.  I also found that some slot combinations outputs the SHA but not the public cert.

scarolan, 2022-07-26:
Here's an example where it sort of works but never outputs the cert

```
Would you like to generate a new private key? (Y/n) [Y]: n

Self-signed certificate is below:

SHA1: da39a3ee5e6b4b0d3255bfef95601890afd80709
```

pennam, 2023-09-18:
@scarolan it looks like you are not using [this](https://github.com/arduino-libraries/ArduinoECCX08/blob/master/examples/Tools/ECCX08SelfSignedCert/ECCX08SelfSignedCert.ino) example to generate the CSR. 

Can you share your sketch and your ATECC608 configuration?

scarolan, 2023-09-25:
Sorry, I've long since abandoned this project.  I think we can close this issue if nobody else cares or is affected.

closed 2024-01-28 by scarolan

VinayBharambe, 2024-10-21:
Same issue of "Error starting self signed cert generation!".

I am using Arduino Due and used 0 and 9 slots.