arduino-libraries/ArduinoECCX08 #66: `-Wvla` compiler warnings

zfields, 2024-07-18:
```
~/Development/Arduino/libraries/ArduinoECCX08/src/utility/ECCX08JWS.cpp: In member function 'arduino::String ECCX08JWSClass::publicKey(int, bool)':
~/Development/Arduino/libraries/ArduinoECCX08/src/utility/ECCX08JWS.cpp:94:18: warning: variable length array 'out' is used [-Wvla]
   byte out[length];
                  ^
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-deprecated-copy'
~/Development/Arduino/libraries/ArduinoECCX08/src/utility/ECCX08SelfSignedCert.cpp: In member function 'int ECCX08SelfSignedCertClass::buildCert(bool)':
~/Development/Arduino/libraries/ArduinoECCX08/src/utility/ECCX08SelfSignedCert.cpp:235:51: warning: variable length array 'certInfo' is used [-Wvla]
   uint8_t certInfo[certInfoLen + certInfoHeaderLen];
                                                   ^
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-deprecated-copy'
~/Development/Arduino/libraries/ArduinoECCX08/src/ECCX08.cpp: In member function 'int ECCX08Class::sendCommand(uint8_t, uint8_t, uint16_t, const byte*, size_t)':
~/Development/Arduino/libraries/ArduinoECCX08/src/ECCX08.cpp:885:29: warning: variable length array 'command' is used [-Wvla]
   byte command[commandLength];
                             ^
~/Development/Arduino/libraries/ArduinoECCX08/src/ECCX08.cpp: In member function 'int ECCX08Class::receiveResponse(void*, size_t)':
~/Development/Arduino/libraries/ArduinoECCX08/src/ECCX08.cpp:910:35: warning: variable length array 'responseBuffer' is used [-Wvla]
   byte responseBuffer[responseSize];
                                   ^
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-deprecated-copy'
~/Development/Arduino/libraries/ArduinoECCX08/src/utility/ECCX08CSR.cpp: In member function 'arduino::String ECCX08CSRClass::end()':
~/Development/Arduino/libraries/ArduinoECCX08/src/utility/ECCX08CSR.cpp:68:45: warning: variable length array 'csrInfo' is used [-Wvla]
   byte csrInfo[csrInfoHeaderLen + csrInfoLen];
                                             ^
~/Development/Arduino/libraries/ArduinoECCX08/src/utility/ECCX08CSR.cpp:130:33: warning: variable length array 'csr' is used [-Wvla]
   byte csr[csrLen + csrHeaderLen];
                                 ^
```

zfields, 2024-07-18:
arduino-cli compile --verbose --warnings=all --build-property "build.extra_flags=-Wno-unused-variable -Wno-implicit-fallthrough -Wno-psabi -Wno-deprecated-copy -Wno-missing-field-initializers -Wno-delete-non-virtual-dtor -Wno-sign-compare -Og -g" --fqbn arduino:mbed_opta:opta <dev folder>

schnoberts1, 2024-11-23:
It makes me sad I have to do the same sort of thing as you to get a clean builder, particularly because of of the things that get suppressed are aliasing violations and other undefined behaviour which often become crashes in later versions of the compiler.